mr-manuel/venus-os_dbus-mqtt-pv #1: Help with Topic

PhileMC, 2023-01-12:
Sorry, can you look at the debug!?

I think i have a Connection, ist there only a problem with the Topic?

[image]

[image]

[image]

mr-manuel, 2023-01-13:
Hello,

if you look at the [minimum required JSON structure](https://github.com/mr-manuel/venus-os_dbus-mqtt-pv#json-structure) you need a JSON file. In your case you are passing a value to a subtopic.

In Node-RED you first have to create an object, then convert it to a JSON structure and then you can pass it to the MQTT topic.

Here is a flow that you can import and should work for you:
```json
[
    {
        "id": "a7dd93e52bb0f458",
        "type": "inject",
        "z": "42e5fe8b46267f4b",
        "name": "",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "5",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "1000",
        "payloadType": "num",
        "x": 710,
        "y": 340,
        "wires": [
            [
                "fca9bbab075a9b7b",
                "c9a75642a0b36957"
            ]
        ]
    },
    {
        "id": "fca9bbab075a9b7b",
        "type": "function",
        "z": "42e5fe8b46267f4b",
        "name": "create JSON",
        "func": "msg.payload = {\n    \"pv\": {\n        \"power\": msg.payload\n    }\n}\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 890,
        "y": 340,
        "wires": [
            [
                "56297b41ff0f8694"
            ]
        ]
    },
    {
        "id": "56297b41ff0f8694",
        "type": "json",
        "z": "42e5fe8b46267f4b",
        "name": "",
        "property": "payload",
        "action": "str",
        "pretty": false,
        "x": 1130,
        "y": 340,
        "wires": [
            [
                "6867f9d68c12002f"
            ]
        ]
    },
    {
        "id": "6867f9d68c12002f",
        "type": "mqtt out",
        "z": "42e5fe8b46267f4b",
        "name": "",
        "topic": "064-KG-Heizung-VictronGX",
        "qos": "",
        "retain": "",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "",
        "x": 1360,
        "y": 340,
        "wires": []
    },
    {
        "id": "c9a75642a0b36957",
        "type": "function",
        "z": "42e5fe8b46267f4b",
        "name": "create JSON with L1",
        "func": "msg.payload = {\n    \"pv\": {\n        \"power\": msg.payload,\n        \"L1\": {\n            \"power\": msg.payload\n        }\n    }\n}\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 920,
        "y": 380,
        "wires": [
            [
                "56297b41ff0f8694"
            ]
        ]
    }
]
```

PhileMC, 2023-01-13:
Hi,
thanks a lot. I will try it later and give you a feedback.

PhileMC, 2023-01-13:
[image]

Thanks a lot.

closed 2023-01-13 by PhileMC

PhileMC, 2023-01-13:
hi, is it possible to "Rename" L1 L2 and L3 to e.g. South; West and East?