mr-manuel/venus-os_dbus-mqtt-pv #5: No MQTT Inverter on VenusOS

Zahnpasta83, 2023-05-29:
Hello, I installed the tool as described in the instructions but I don't get a mqtt inverter displayed in venusos. 
Debug was carried out and the tool started without any problems. 
Data is sent via the mqtt topic enphase/envoy-s/meters. do i have to do anything else to get the tool to work?

Zahnpasta83, 2023-05-29:
when looking in the logs i get following message "ERROR:root:Exception occurred: argument of type 'int' is not iterable"

mr-manuel, 2023-05-29:
Hello, please post the complete log since driver starting and the published JSON.

Zahnpasta83, 2023-05-29:
Ich hab es irgendwie fertig gebracht das ein MQTT Inverter in Venus auftaucht. Allerdings bin ich offensichtlich nicht in der Lage die Werte die ich in NodeRed bekomme vernünftig in ein JSON Format zu konvertieren. Vielleicht könntest du mir da aushelfen. Ich bekomme reine Zahlenwerte und habe dort eine Change Node dran gehängt. Wie muss ich diese konfigurieren [image]

mr-manuel, 2023-05-29:
Which values do you have available? Only the PV power? Please share the whole flow.

Zahnpasta83, 2023-05-29:
A lot of trail and error but it's working now [image]

[image]

[image]

closed 2023-05-29 by mr-manuel