mr-manuel/venus-os_dbus-mqtt-pv #6: Help with the correct 3 Phase config

blackpearlla, 2023-06-14:
Hi,

I'm probably too stupid to map the configuration for 3 phases in Node RED. can someone please help me?

Here is my test configuration.

`[
    {
        "id": "3e7d5456fd2222e2",
        "type": "json",
        "z": "c0c1fa74bf2f0e98",
        "name": "",
        "property": "payload",
        "action": "str",
        "pretty": false,
        "x": 970,
        "y": 340,
        "wires": [
            [
                "603843446f084cf1"
            ]
        ]
    },
    {
        "id": "57e3f70fdb658f06",
        "type": "function",
        "z": "c0c1fa74bf2f0e98",
        "name": "create JSON with L1",
        "func": "msg.payload = {\n    \"pv\": {\n        \"power\": msg.payload,\n        \"L1\": {\n            \"power\": msg.payload\n        }\n    }\n}\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 300,
        "y": 300,
        "wires": [
            [
                "6fb7559d6c246f91",
                "d15ac975819c2098"
            ]
        ]
    },
    {
        "id": "a47b1adfe30863ee",
        "type": "function",
        "z": "c0c1fa74bf2f0e98",
        "name": "create JSON with L2",
        "func": "msg.payload = {\n    \"pv\": {\n        \"power\": msg.payload,\n        \"L2\": {\n            \"power\": msg.payload\n        }\n    }\n}\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 300,
        "y": 340,
        "wires": [
            [
                "6fb7559d6c246f91",
                "d15ac975819c2098"
            ]
        ]
    },
    {
        "id": "773c74089e51aa80",
        "type": "function",
        "z": "c0c1fa74bf2f0e98",
        "name": "create JSON with L3",
        "func": "msg.payload = {\n    \"pv\": {\n        \"power\": msg.payload,\n        \"L3\": {\n            \"power\": msg.payload\n        }\n    }\n}\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 300,
        "y": 380,
        "wires": [
            [
                "6fb7559d6c246f91",
                "d15ac975819c2098"
            ]
        ]
    },
    {
        "id": "40ed6cfcca33afec",
        "type": "mqtt in",
        "z": "c0c1fa74bf2f0e98",
        "name": "PV Power L1",
        "topic": "homeassistant/sensor/on_grid_l1_power/state",
        "qos": "2",
        "datatype": "auto-detect",
        "broker": "e72e968db75c5c1e",
        "nl": false,
        "rap": true,
        "rh": 0,
        "inputs": 0,
        "x": 90,
        "y": 300,
        "wires": [
            [
                "57e3f70fdb658f06",
                "91098df261fd22bb"
            ]
        ]
    },
    {
        "id": "58c9bf31b2bc9c45",
        "type": "mqtt in",
        "z": "c0c1fa74bf2f0e98",
        "name": "PV Power L2",
        "topic": "homeassistant/sensor/on_grid_l2_power/state",
        "qos": "2",
        "datatype": "auto-detect",
        "broker": "e72e968db75c5c1e",
        "nl": false,
        "rap": true,
        "rh": 0,
        "inputs": 0,
        "x": 90,
        "y": 340,
        "wires": [
            [
                "a47b1adfe30863ee",
                "91098df261fd22bb"
            ]
        ]
    },
    {
        "id": "e4b9b5ae41499d92",
        "type": "mqtt in",
        "z": "c0c1fa74bf2f0e98",
        "name": "PV Power L3",
        "topic": "homeassistant/sensor/on_grid_l3_power/state",
        "qos": "2",
        "datatype": "auto-detect",
        "broker": "e72e968db75c5c1e",
        "nl": false,
        "rap": true,
        "rh": 0,
        "inputs": 0,
        "x": 90,
        "y": 380,
        "wires": [
            [
                "773c74089e51aa80",
                "91098df261fd22bb"
            ]
        ]
    },
    {
        "id": "603843446f084cf1",
        "type": "debug",
        "z": "c0c1fa74bf2f0e98",
        "name": "debug Payload",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 1160,
        "y": 260,
        "wires": []
    },
    {
        "id": "6fb7559d6c246f91",
        "type": "function",
        "z": "c0c1fa74bf2f0e98",
        "name": "wenn Wert nicht verfügbar",
        "func": "if (msg.payload && msg.payload.pv && msg.payload.pv.power === \"unavailable\" && msg.payload.pv.L1 && msg.payload.pv.L1.power === \"unavailable\") {\n    msg.payload = {\n        \"pv\": {\n            \"power\": \"0\",\n            \"L1\": {\n                \"power\": \"0\"\n            }\n        }\n    };\n}\nreturn msg;\n",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 630,
        "y": 340,
        "wires": [
            [
                "3e7d5456fd2222e2",
                "96af2d1fcaf822bf"
            ]
        ]
    },
    {
        "id": "96af2d1fcaf822bf",
        "type": "debug",
        "z": "c0c1fa74bf2f0e98",
        "name": "debug Power State \"unknown\"",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 890,
        "y": 260,
        "wires": []
    },
    {
        "id": "d15ac975819c2098",
        "type": "debug",
        "z": "c0c1fa74bf2f0e98",
        "name": "debug create JSON",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 610,
        "y": 260,
        "wires": []
    },
    {
        "id": "91098df261fd22bb",
        "type": "debug",
        "z": "c0c1fa74bf2f0e98",
        "name": "debug MQTT request",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 300,
        "y": 260,
        "wires": []
    },
    {
        "id": "e72e968db75c5c1e",
        "type": "mqtt-broker",
        "name": "Homeassistant MQTT Broker",
        "broker": "192.168.178.126",
        "port": "1883",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "5",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": ""
    }
]`

mr-manuel, 2023-06-14:
Try this:

```json
[
    {
        "id": "3e7d5456fd2222e2",
        "type": "json",
        "z": "42e5fe8b46267f4b",
        "name": "",
        "property": "payload",
        "action": "str",
        "pretty": false,
        "x": 930,
        "y": 260,
        "wires": [
            [
                "f709e6f07c8df0ec"
            ]
        ]
    },
    {
        "id": "57e3f70fdb658f06",
        "type": "function",
        "z": "42e5fe8b46267f4b",
        "name": "L1",
        "func": "msg.topic = \"L1\"\n\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 330,
        "y": 220,
        "wires": [
            [
                "b0dd0b4d12033e09"
            ]
        ]
    },
    {
        "id": "40ed6cfcca33afec",
        "type": "mqtt in",
        "z": "42e5fe8b46267f4b",
        "name": "PV Power L1",
        "topic": "homeassistant/sensor/on_grid_l1_power/state",
        "qos": "2",
        "datatype": "auto-detect",
        "broker": "e72e968db75c5c1e",
        "nl": false,
        "rap": true,
        "rh": 0,
        "inputs": 0,
        "x": 170,
        "y": 220,
        "wires": [
            [
                "57e3f70fdb658f06"
            ]
        ]
    },
    {
        "id": "58c9bf31b2bc9c45",
        "type": "mqtt in",
        "z": "42e5fe8b46267f4b",
        "name": "PV Power L2",
        "topic": "homeassistant/sensor/on_grid_l2_power/state",
        "qos": "2",
        "datatype": "auto-detect",
        "broker": "e72e968db75c5c1e",
        "nl": false,
[... 166 lines cut ...]
        "broker": "192.168.178.185",
        "port": "1883",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "5",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": ""
    }
]
```

blackpearlla, 2023-06-15:
Thanks a lot. It seems to work!

closed 2023-06-15 by blackpearlla