mr-manuel/venus-os_dbus-mqtt-pv #8: possible installation without mqtt config ?

TTr0n, 2023-07-10:
Hi, nice idea! :-)
Is it possible to use this version, to register a pv Inverter in Venus Os without taking values from MQTT ?
I am able to send the PV Inverter data via modbus direct to the adress in VenuOs.

Greetings

mr-manuel, 2023-07-10:
Thats not possible, but you can use Node-RED (included in the Venus OS large version) to fetch the data from Modbus and send it via MQTT to the broker.

TTr0n, 2023-07-11:
Ich habe mich schon ein wenig mit dem NodeRed beschäftigt, allerdings funktioniert das doch erst, wenn der Dienst/PV Inverter dem System bekannt gemacht wurde und das geht doch nur, wenn Dein Script sauber durchläuft, oder habe ich da einen Denkfehler?
Evtl. habe ich mich auch falsch ausgedrückt. Der Plan war die Daten per Modbus ans Venus zu schicken und dann wie von Dir vorgeschlagen per MQTT weiter zu leiten, nur fehlt mir da der Ansatz, wie genau das funktioniert. Ein entsprechendes Tutorial oder Video habe ich bisher nicht finden können.

mr-manuel, 2023-07-11:
Venus OS will not recognize the Modbus data unless the device is supported. You have to convert your data therefore first to MQTT, else it won't work. This has to be done by other software than this driver. One way to go could be:

Your device -> Modbus -> Node-RED -> MQTT -> dbus-mqtt-pv -> Venus OS

I don't know how you thought about it, but MQTT is the core part that is needed for this driver. It won't run without it.

You can also take a look at [modpoll](https://github.com/gavinying/modpoll). Which is a separate programm to fetch data from Modbus and send it to MQTT.

Google first on how to get the data from Modbus into Node-RED. Once you are able to it's easy to send it to MQTT.

closed 2023-08-06 by mr-manuel