mr-manuel/venus-os_dbus-mqtt-pv #9: float is not iterable

tritechks, 2023-07-27:
Hello, after installation i became this error message (see below). I try to send data from my trucki2shelly gateway information to emulate my SUN1000G2 as a Victron Inverter.

@4000000064c258cb15b4444c DEBUG:root:MQTT payload: '194.40'
@4000000064c258cd059976cc ERROR:root:Exception occurred: argument of type 'float' is not iterable

Greetings Thomas

mr-manuel, 2023-07-27:
Did you check the [README.md](https://github.com/mr-manuel/venus-os_dbus-mqtt-pv#json-structure)? From the log it seems, that you are not passing a JSON payload via MQTT but only a float number.

mr-manuel, 2023-07-28:
Unfortunately I don't know the product you are using, but you can use Node-RED to read and convert the payload. Therefore install the Venus OS large firmware. You can select it in Venus OS remote console/GUI under Firmware.

closed 2023-08-22 by mr-manuel