mr-manuel/venus-os_dbus-mqtt-pv #10: PV inverter not shown

wasn-eu, 2023-08-06:
Hello,

i installed your dbus-mqtt tools to my raspberry pi running venus OS large v3.0.1
all other implementations are shown but not the pv inverter.
in the setting i can see it:
[image]

But in the graphical overview it is not shown:
[image]

please help

mr-manuel, 2023-08-06:
Try to change this:

https://github.com/mr-manuel/venus-os_dbus-mqtt-pv/blob/a3af5ae1d0372fb9d3859a1dd6d2ee301937b59c/dbus-mqtt-pv/config.sample.ini#L34-L38

wasn-eu, 2023-08-06:
that position=0 is the default
i changed it to position=1 and now it ist shown.

thank you for the fast help

closed 2023-08-06 by wasn-eu