mr-manuel/venus-os_dbus-mqtt-pv #11: wrong calaculation

wasn-eu, 2023-08-06:
Hello

why show the ac load widget the same value as the pv inverter widget when there is a negative powef in ac input?
[image]

the ac loads power value of the  phase should be pv inverter value + ac input value
for example: on phase 3
  pv inverter power: 340W
  ac input power: -67W
that should be:
  ac loads: 273W

because i send 67W back to the grid

wasn-eu, 2023-08-06:
if i have a positive power for L3 the calculation ist correct:
[image]

the pv inverter power and ac input power in sum is the ac loads power

mr-manuel, 2023-08-06:
The `dbus-mqtt-pv` does not make any calculations. This is because of the `dbus-systemcalc-py` from Victron and the fact that you don't have a MultiPlus.
Try to use the [`dbus-multiplus-emulator`](https://github.com/mr-manuel/venus-os_dbus-multiplus-emulator) from my repository.

wasn-eu, 2023-08-06:
i have now installed the multiplus-emulator and get this:
[image]

the AC input on L3 is correct but the AC loads on L3 is allways 0 it should be: grid l3 -  pv l3
can i configure the emulator for that?

wasn-eu, 2023-08-06:
perhabs you can disable the critital loads?

mr-manuel, 2023-08-06:
Try to change the position of the PV to `0` again.

https://github.com/mr-manuel/venus-os_dbus-mqtt-pv/blob/a3af5ae1d0372fb9d3859a1dd6d2ee301937b59c/dbus-mqtt-pv/config.sample.ini#L34-L38

wasn-eu, 2023-08-06:
it is now on position=0 but without a change:
looks a little different cause i just installed the gui mods:

[image]

wasn-eu, 2023-08-06:
it must be the emulator.
when i uninstall it it get this:
[image]

this is right.
the sum of grid L3 and PV L3 are the Loads L3

mr-manuel, 2023-08-06:
With GuiMods it gets displayed in another way. So now it‘s displayed correctly?

wasn-eu, 2023-08-06:
yes now it is correct

closed 2023-08-06 by wasn-eu