mr-manuel/venus-os_dbus-mqtt-pv #13: Why timeout after 60s with no mqtt messages ?

gdave321, 2023-08-22:
According to the source, after receiving no new mqtt messages after 60s (default)  the driver will exit.  I assume that will cause the driver to restart since it's a linux service.
However, a pv-inverters will typically not produce any data during night times and having the driver restart (and eventually die?) after hours and hours doesn't seem to make much sense.  So what am I missing here?

mr-manuel, 2023-08-22:
Because this is the Victron standard, if there is no new message/data. Else you also don't see, if your system is working properly or not.

If you don't want to use this, you can disable the timeout in the `config.ini` or set it to `86400` (1 day), so the driver exits after one day, if there is no new data.

gdave321, 2023-08-22:
Ok, clear. Thanks for the update.

gdave321, 2023-08-22:
Close.

closed 2023-08-22 by gdave321