mr-manuel/venus-os_dbus-mqtt-pv #14: JSON Syntax

CharlieBerghammer, 2023-08-26:
Hallo Manuel, 
habe längere Zeit gesucht, wie ich meine älteren Wechselrichter in Victron integrieren kann. Hier ist dein 
dbus-mqtt-pv Adapter super. Vielen Dank dafür.
Meine PV Daten bekomme ich per http get. Ich mache hier in Node Red 4 Abfragen für PV Leistung "Gesamt" "L1" "L2" und "L3". Dann nehme ich die Funktion "change", wandle den Wert in das JSON Format und schicke es mit "mqtt out" wieder weiter. Das Readme habe ich gelesen, komme aber nicht weiter.
Mit der PV Gesamtleistung funktioniert alles bestens. In Node Red gebe ich für das JSON Format 
value JSON      {"pv": { "power":payload } }   und alles funktioniert perfekt.
Möchte ich aber auch L1 L2 L3 anzeigen lassen, weiß ich nicht wie ich das Ganze ins JSON Format bringen kann. Beim Versuch 
{"pv": {"power": 0.0, "L1": {"power": 0.0 }, "L2": {"power": 0.0 }, "L3": {"power": 0.0 }}} einzubinden komme ich nicht weiter. Kannst Du mir hier bitte einen Tip geben wie ich hier weiter komme.
Grüße Charlie

mr-manuel, 2023-08-26:
Hallo Charlie,
wenn ich es richtig verstanden habe, dann weißt du nicht wie du die vier Werte der getrennten Abfragen zusammenzufassen kannst und als eine JSON auszugeben kannst?
Ich würde die vier Ausgaben in einer Flow Variable speichern und mit einem Injection und Function Node die Werte alle 5 Sekunden (oder wie benötigt/abgefragt) ausgeben.
Kannst du den Flow den du bis jetzt hast als JSON hier posten?

CharlieBerghammer, 2023-08-26:
Hallo Manuel,
Ich habe leider von Node Red wenig und von JSON gar keine Ahnung, Ich poste mal was ich bis jetzt vergeblich gemacht habe.
[
    {
        "id": "8e44be37516623b7",
        "type": "mqtt out",
        "z": "bb3cd8119f4b05ce",
        "name": "Power",
        "topic": "Victron-PV/power",
        "qos": "",
        "retain": "",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "9b344c22b0d918b9",
        "x": 850,
        "y": 80,
        "wires": []
    },
    {
        "id": "1e7173c2e8662ce5",
        "type": "change",
        "z": "bb3cd8119f4b05ce",
        "name": "Gesamtleistung",
        "rules": [
            {
                "t": "set",
                "p": "payload",
                "pt": "msg",
                "to": "{\"pv\": { \"power\":payload } }",
                "tot": "jsonata"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 540,
        "y": 80,
        "wires": [
            [
                "8e44be37516623b7"
            ]
        ]
    },
    {
        "id": "3cb3062cf062eed5",
        "type": "http request",
        "z": "bb3cd8119f4b05ce",
        "name": "",
        "method": "GET",
        "ret": "txt",
        "paytoqs": "ignore",
        "url": "http://192.168.2.15:8087/getPlainValue/solarviewdatareader.0.pvi3.current",
        "tls": "",
        "persist": true,
        "proxy": "",
        "insecureHTTPParser": true,
        "authType": "",
        "senderr": false,
        "headers": [],
        "x": 350,
        "y": 140,
        "wires": [
            [
                "8e8ae80541a3b932"
            ]
[... 329 lines cut ...]
        "name": "iobroker",
        "broker": "192.168.2.15",
        "port": "1884",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": ""
    }
]

mr-manuel, 2023-08-27:
Try with this:

```json
[
    {
        "id": "a06b3000d15dbefd",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "8e44be37516623b7",
        "type": "mqtt out",
        "z": "a06b3000d15dbefd",
        "name": "Venus OS - PV Power",
        "topic": "Victron-PV/power",
        "qos": "",
        "retain": "",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "9b344c22b0d918b9",
        "x": 860,
        "y": 180,
        "wires": []
    },
    {
        "id": "3cb3062cf062eed5",
        "type": "http request",
        "z": "a06b3000d15dbefd",
        "name": "L3",
        "method": "GET",
        "ret": "txt",
        "paytoqs": "ignore",
        "url": "http://192.168.2.15:8087/getPlainValue/solarviewdatareader.0.pvi3.current",
        "tls": "",
        "persist": true,
        "proxy": "",
        "insecureHTTPParser": true,
        "authType": "",
        "senderr": false,
        "headers": [],
        "x": 350,
        "y": 240,
        "wires": [
            [
                "554dfd7ba5f6effa"
            ]
        ]
    },
    {
        "id": "1d5ffcc0a1dd435a",
        "type": "http request",
        "z": "a06b3000d15dbefd",
        "name": "L2",
        "method": "GET",
        "ret": "txt",
        "paytoqs": "ignore",
        "url": "http://192.168.2.15:8087/getPlainValue/solarviewdatareader.0.pvi2.current",
        "tls": "",
        "persist": false,
        "proxy": "",
        "insecureHTTPParser": false,
        "authType": "",
        "senderr": false,
        "headers": [],
[... 229 lines cut ...]
        "broker": "192.168.2.15",
        "port": "1884",
        "clientid": "",
        "autoConnect": true,
        "usetls": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": ""
    }
]
```

CharlieBerghammer, 2023-08-27:
Vielen Dank Manuel ,
das hat mir, auch für mein Verständnis wirklich weitergeholfen.
In dem Flow ist ein kleiner Fehler, denn die Verbindung von L1 "Set topic" zu "join" fehlte und damit ist nach "join" nichts rausgekommen.
Sehe jetzt alle Daten :-). Habe jetzt nur noch ein Problem mit der grafischen Anzeige der Victron Console. Hier sehe ich nur den Gesamtwert und nicht die einzelnen Phasen. In der Geräteliste unter MQTT PV sind aber alle Phasen vorhanden. Habe ich hier in der config.ini noch etwas übersehen?

[image]
[image]

mr-manuel, 2023-08-27:
Das ist richtig so. Wenn du alle Phasen sehen willst musst du die [GuiMods](https://github.com/kwindrem/GuiMods) installieren.

closed 2023-08-27 by mr-manuel