mr-manuel/venus-os_dbus-mqtt-pv #15: Treiber autostart

Jacke87, 2023-08-27:
Hallo zusammen,
erstmal eines Vorweg der Treiber funktioniert super. Nur nach einem neustart startet der Treiber nicht. so sieht meine RC Local aus
`bash /data/etc/dbus-serialbattery/reinstall-local.sh
(cd /data/venus_kostal_pico/ && python kostal.py)
bash /data/etc/dbus-mqtt-pv/install.sh
`
gebe ich nach dem neustart 
`bash /data/etc/dbus-mqtt-pv/install.sh`
ein funktioniert es.

`chmod +x /data/rc.local`
Habe ich auch schon versucht und der Rest startet

mr-manuel, 2023-08-27:
Setzte das `bash /data/etc/dbus-mqtt-pv/install.sh` in der `rc.local` mal an die 1. oder 2. Stelle.
Könnte sein, dass das Kostal Script nicht korrekt durchläuft und deshalb die Scripts dannach nicht ausgeführt werden.

Jacke87, 2023-08-30:
Hey super das war es vielen Dank.
Mal eine Frage ich habe keine Ahnung wo es hingehört oder ob es hier reingehört.
Ich habe folgende konfiguration
Raspberry Venus os - IoBroker - Node RED auf IoBroker - Growatt mit modstick mit ESP Home 
Ich sende die werte des Growatt an den IoBroker von da dann mit Node Red an da VenusOS.
Nun habe ich das ESHome skript so angepasst das es direkt die Werte an den Treiber übergibt.
So ist es fast eine Native Verbindung.
Könnte so etwas unter Projekte aufgeführt werden?
Nur eine Sache ist noch nicht sehr schön der IoBroker hat bei mir noch die funktion des Brokers geht es irgendwie ohne den, z.b den Broker des Venus zu verwenden?

mr-manuel, 2023-08-30:
Wie meinst du das mit unter Projekte aufführen?

Wenn du das ESPHome Script so anpasst, dass es das benötigte Format hat und du den MQTT Broker von Venus OS nutzt, dann hast du so wenig Komponenten wie nur möglich. Der Venus OS Broker ist ein normaler Broker wie jeder andere. Da suchst du dir am besten einfach ein noch nicht verwendeted Topic aus.

Jacke87, 2023-09-02:
Also es läuft jetzt komplett ohne Zwischenstation vielen Dank. 
Als Beispiel quasi mit Anleitung. Falls jemand etwas ähnliches hat.
Habe einige github Projekte gesehen wo Beispiele im Wiki mit drin waren?

closed 2023-09-02 by Jacke87

mr-manuel, 2023-09-03:
Achso. Ja es gibt so viele verschiedene Möglichkeiten, da sollte jeder sein eigenes System kennen. Gerne kannst du dein Beispiel hier posten, dann füge ich es ein.