mr-manuel/venus-os_dbus-mqtt-pv #16: PV Inverter wird nicht im VRM Portal angezeigt

fl0wb0b, 2023-10-26:
Hallo ich habe Position=0 in der config ausgewählt, jedoch wird er nicht angezeigt. FW= 3.11
[image]
[image]

stelle ich die Position auf 1, wird im Portal falsch gerechnet.
[image]

fl0wb0b, 2023-10-26:
ich habe den Debug angemacht und schaue in die logs:

2023-10-19 19:51:19.168411500 *** starting dbus-mqtt-grid ***
2023-10-19 19:51:19.168975500 ./run: line 4: /data/etc/dbus-mqtt-grid/dbus-mqtt-grid.py: No such file or directory
2023-10-19 19:51:20.193057500 *** starting dbus-mqtt-grid ***
2023-10-19 19:51:20.193717500 ./run: line 4: /data/etc/dbus-mqtt-grid/dbus-mqtt-grid.py: No such file or directory
2023-10-19 19:51:21.215949500 *** starting dbus-mqtt-grid ***
2023-10-19 19:51:21.216506500 ./run: line 4: /data/etc/dbus-mqtt-grid/dbus-mqtt-grid.py: No such file or directory

Kann mir jemand helfen? :-S

mr-manuel, 2023-10-27:
How is your system build?

fl0wb0b, 2023-10-27:
Raspberry Pi Venus OS 3.11
EM540 as gridmeter
no Multiplus or Quattro
SMA Modbus -> NodeRed -> MQTT (local)

Which version of Venus OS are you using?

mr-manuel, 2023-10-27:
Can you please describe more in detail how your system looks like?

1. Which devices are you using (make, model, voltage, power)?
2. How is your system cabled?
3. Are you using one or three phase system? Where is the PV Inverter connected to (should be covered from 2.)?

The PV Inverter (I assume 230 V, since you wrote in german) won't work if you have no other 230 V component in the system.

Could it be that you only have a 230 V third party PV inverter and want to monitor it with Victron?

fl0wb0b, 2023-10-27:
I have installed the EM540 near my power connection. There is only one SMA Tripower X 12, which has three phases L1-L3. This is of course located behind the gridmeter. At first I wanted to connect the inverter to the Raspberry Pi via Modbus, but no inverter is detected. Thereupon I built a flow in Node Red. The data is output at the end as follows: {"pv":{"power":526,"voltage":230,"current":3.502,"energy_forward":30.68,"L1":{"power":177,"voltage":230.11,"current":1.162},"L2":{"power":174,"voltage":229.26,"current":1.165},"L3":{"power":175,"voltage":230.95,"current":1.176}}}
[image]

config:

> ; CONFIG FILE
> ; GitHub reporitory: https://github.com/mr-manuel/venus-os_dbus-mqtt-pv
> ; remove semicolon ; to enable desired setting
> 
> [DEFAULT]
> ; Set logging level
> ; ERROR = shows errors only
> ; WARNING = shows ERROR and warnings
> ; INFO = shows WARNING and running functions
> ; DEBUG = shows INFO and data/values
> ; default: WARNING
> logging = DEBUG
> 
> ; Device name
> ; default: MQTT PV
> device_name = SMA-STP-X-12
> 
> ; Device VRM instance
> ; default: 51
> device_instance = 51
> 
> ; Specify after how many seconds the driver should exit (disconnect), if no new MQTT message was received
> ; default: 60
> ; value to disable timeout: 0
> timeout = 60
> 
> ; used when no voltage is received
> voltage = 230
> 
> 
> [PV]
> ; Max rated power (in Watts) of the inverter
> max = 12000
> ; position (connected input/output) of the PV inverter
> ; 0 = AC input 1 (select this if you don't have a MultiPlus or Quattro)
> ; 1 = AC output
> ; 2 = AC input 2
> position = 0
> 
> 
> [MQTT]
> ; IP addess or FQDN from MQTT server
> broker_address = localhost
> 
> ; Port of the MQTT server
> ; default plaintext: 1883
> ; default TLS port: 8883
> broker_port = 1883
> 
> ; Enables TLS
> ; 0 = Disabled
> ; 1 = Enabled
> ;tls_enabled = 1
> 
> ; Absolute path to the Certificate Authority certificate file that is to be treated as trusted by this client
> ;tls_path_to_ca = /data/keys/mosquitto.crt
> 
> ; Disables verification of the server hostname in the server certificate
> ; 0 = Disabled
> ; 1 = Enabled
> ;tls_insecure = 1
> 
> ; Username used for connection
> ;username = myuser
> 
> ; Password used for connection
> ;password = mypassword
> 
> ; Topic where the pv data as JSON string is published
> ; minimum required JSON payload: {"pv": { "power": 0.0 } }
> topic = sma
> 

Service are running:
root@raspberrypi2:/data/etc/dbus-mqtt-pv# svstat /service/dbus-mqtt-pv
/service/dbus-mqtt-pv: up (pid 10849) 45150 seconds

fl0wb0b, 2023-10-27:
[image]
The inverter is only not displayed in the dashboard (overview), everything else seems to work

fl0wb0b, 2023-10-27:
I could not find the problem but you wrote a great multiplus emlulator that I installed. Now the PV inverter is also displayed on ACIN. For me the problem is solved.
thanks!
[image]

mr-manuel, 2023-10-27:
The system will not display the PV Inverter without a MutliPlus or Quattro. Exactly therefore I write the MutliPlus Emulator ;)

closed 2023-10-27 by mr-manuel