mr-manuel/venus-os_dbus-mqtt-pv #17: From the Victron docs, a running PV inverter should have status 7 (running), not 0 (startup)

MC68030, 2023-11-05:
Hello,
just a small remark, according to the victron dbus docs, for PV inverters, the status message for a running PV inverter should have 7;  0 means startup.
Maybe within the disconnect block, you can also send status 10 (== error)

And thank for the project, it helped my to get my PV Inverter values to the victron world (had to adapt some fixed JSON strings to my MQTT message formats, but not a big deal)

mr-manuel, 2023-11-05:
Somehow I commented it out

https://github.com/mr-manuel/venus-os_dbus-mqtt-pv/blob/a3af5ae1d0372fb9d3859a1dd6d2ee301937b59c/dbus-mqtt-pv/dbus-mqtt-pv.py#L262

Thanks

mr-manuel, 2023-11-06:
I rechecked the situation. The `/StatusCode` and `/ErrorCode` do not change nothing. They are only displayed for Fronius inverters and do not trigger any action.

`/opt/victronenergy/gui/qml/PageAcInModelDefault.qml`

```qml
	MbItemOptions {
		description: qsTr("Status")
		bind: productIdItem.value === froniusInverterProductId ? service.path("/StatusCode") : ""
		readonly: true
		show: productIdItem.value === froniusInverterProductId
		possibleValues: [
			MbOption { description: formatStatus(qsTr("Startup"), 0); value: 0 },
			MbOption { description: formatStatus(qsTr("Startup"), 1); value: 1 },
			MbOption { description: formatStatus(qsTr("Startup"), 2); value: 2 },
			MbOption { description: formatStatus(qsTr("Startup"), 3); value: 3 },
			MbOption { description: formatStatus(qsTr("Startup"), 4); value: 4 },
			MbOption { description: formatStatus(qsTr("Startup"), 5); value: 5 },
			MbOption { description: formatStatus(qsTr("Startup"), 6); value: 6 },
			MbOption { description: qsTr("Running"); value: 7 },
			MbOption { description: qsTr("Standby"); value: 8 },
			MbOption { description: qsTr("Boot loading"); value: 9 },
			MbOption { description: qsTr("Error"); value: 10 },
			MbOption { description: qsTr("Running (MPPT)"); value: 11 },
			MbOption { description: qsTr("Running (Throttled)"); value: 12 }
		]
	}

	MbItemValue {
		description: qsTr("Error Code")
		item.bind: show ? service.path("/ErrorCode") : ""
		show: productIdItem.value === froniusInverterProductId
	}
```

`froniusInverterProductId ` is defined with `0xA142`.

closed 2023-11-06 by mr-manuel

MC68030, 2023-11-07:
Hi,
at least these status codes are used inside VRM Portal, under advanced, there is a widget for PV inverter status available, and it will display as headline the status: starting (or running, if you push 7 as code).  Together with total energy etc.

KR

mr-manuel, 2023-11-07:
AS you maybe saw I neverthless added the status codes :-)

commit b68a29d1 (2023-11-06): added status code, fixes #17