mr-manuel/venus-os_dbus-mqtt-pv #18: fehlende Werte

baetzst, 2023-11-07:
Hi, mein JSON string schaut folgendermaßen aus:
`{"pv": {"power": 1793,"energy_forward": 129942.80817195191,"voltage": 233.78666144112745,"L1": {"power": 597.6666666666666,"voltage": 234.55999475717545},"L2": {"power": 597.6666666666666,"voltage": 233.0399947911501},"L3": {"power": 597.6666666666666,"voltage": 233.75999477505684}}}`

In der Anzeige fehlen aber die Werte für L2 und L3. Es wird nur L1 und gesamt angezeigt:
[image]

Passt da bei meinem JSON was nicht?

mr-manuel, 2023-11-07:
Hello Stefan,

when the driver starts for the first time it checks which phases are present.

1) Are you sure that all phases are sent when the driver starts?
2) What happen if you restart the driver?
3) Have you checked the send value with an MQTT explorer?

baetzst, 2023-11-07:
Here is the complete log after restart:

> 2023-11-07 14:05:31.277267500 INFO:root:Waiting 5 seconds for receiving first data...
2023-11-07 14:05:31.483696500 ERROR:root:Exception occurred: 'frequency'
2023-11-07 14:05:31.484637500 DEBUG:root:MQTT payload: '{"pv": {"power": 222,"energy_forward": 129945,"L1": {"power": 74,"voltage": 233},"L2": {"power": 74,"voltage": 233},"L3": {"power": 74,"voltage": 232}}}'
2023-11-07 14:05:33.345982500 ERROR:root:Exception occurred: 'frequency'
2023-11-07 14:05:33.346738500 DEBUG:root:MQTT payload: '{"pv": {"power": 222,"energy_forward": 129945,"L1": {"power": 74,"voltage": 233},"L2": {"power": 74,"voltage": 233},"L3": {"power": 74,"voltage": 232}}}'
2023-11-07 14:05:36.300868500 DEBUG:root:VeDbusTreeExport / has been created
2023-11-07 14:05:36.310783500 INFO:root:registered ourselves on D-Bus as com.victronenergy.pvinverter.mqtt_pv_51
2023-11-07 14:05:36.312487500 DEBUG:root:com.victronenergy.pvinverter.mqtt_pv_51 /DeviceInstance = 51
2023-11-07 14:05:36.314836500 DEBUG:root:VeDbusTreeExport /Mgmt has been created
2023-11-07 14:05:36.316237500 DEBUG:root:added /Mgmt/ProcessName with start value /data/etc/dbus-mqtt-pv/dbus-mqtt-pv.py. Writeable is False
2023-11-07 14:05:36.318776500 DEBUG:root:added /Mgmt/ProcessVersion with start value Unkown version, and running on Python 3.8.14. Writeable is False
2023-11-07 14:05:36.320207500 DEBUG:root:added /Mgmt/Connection with start value MQTT PV service. Writeable is False
2023-11-07 14:05:36.321608500 DEBUG:root:added /DeviceInstance with start value 51. Writeable is False
2023-11-07 14:05:36.322927500 DEBUG:root:added /ProductId with start value 65535. Writeable is False
2023-11-07 14:05:36.331597500 DEBUG:root:added /ProductName with start value MQTT PV. Writeable is False
2023-11-07 14:05:36.334970500 DEBUG:root:added /CustomName with start value MQTT PV. Writeable is False
2023-11-07 14:05:36.338054500 DEBUG:root:added /FirmwareVersion with start value 0.1.4 (20231106). Writeable is False
2023-11-07 14:05:36.339282500 DEBUG:root:added /Connected with start value 1. Writeable is False
2023-11-07 14:05:36.340448500 DEBUG:root:added /Latency with start value None. Writeable is False
2023-11-07 14:05:36.341786500 DEBUG:root:added /ErrorCode with start value 0. Writeable is False
2023-11-07 14:05:36.343611500 DEBUG:root:added /Position with start value 0. Writeable is False
2023-11-07 14:05:36.345301500 DEBUG:root:added /StatusCode with start value 0. Writeable is False
2023-11-07 14:05:36.349412500 DEBUG:root:VeDbusTreeExport /Ac has been created
2023-11-07 14:05:36.353515500 DEBUG:root:added /Ac/Power with start value 0. Writeable is True
2023-11-07 14:05:36.354911500 DEBUG:root:added /Ac/Current with start value 0. Writeable is True
2023-11-07 14:05:36.355980500 DEBUG:root:added /Ac/Voltage with start value 0. Writeable is True
2023-11-07 14:05:36.357121500 DEBUG:root:VeDbusTreeExport /Ac/Energy has been created
2023-11-07 14:05:36.357805500 DEBUG:root:added /Ac/Energy/Forward with start value None. Writeable is True
2023-11-07 14:05:36.358943500 DEBUG:root:VeDbusTreeExport /Ac/L1 has been created
2023-11-07 14:05:36.359607500 DEBUG:root:added /Ac/L1/Power with start value 0. Writeable is True
2023-11-07 14:05:36.360549500 DEBUG:root:added /Ac/L1/Current with start value 0. Writeable is True
2023-11-07 14:05:36.361487500 DEBUG:root:added /Ac/L1/Voltage with start value 0. Writeable is True
2023-11-07 14:05:36.362559500 DEBUG:root:added /Ac/L1/Frequency with start value None. Writeable is True
2023-11-07 14:05:36.363948500 DEBUG:root:VeDbusTreeExport /Ac/L1/Energy has been created
2023-11-07 14:05:36.364711500 DEBUG:root:added /Ac/L1/Energy/Forward with start value None. Writeable is True
2023-11-07 14:05:36.365728500 DEBUG:root:added /Ac/MaxPower with start value 10000. Writeable is True
2023-11-07 14:05:36.366675500 DEBUG:root:added /Ac/Position with start value 0. Writeable is True
2023-11-07 14:05:36.367612500 DEBUG:root:added /Ac/StatusCode with start value 0. Writeable is True
2023-11-07 14:05:36.368509500 DEBUG:root:added /UpdateIndex with start value 0. Writeable is True
2023-11-07 14:05:36.370706500 INFO:root:Connected to dbus and switching over to GLib.MainLoop() (= event based)
2023-11-07 14:05:36.397220500 DEBUG:root:_get_value_handler called for /
2023-11-07 14:05:36.400796500 DEBUG:root:{'Mgmt/ProcessName': dbus.String('/data/etc/dbus-mqtt-pv/dbus-mqtt-pv.py', variant_level=1), 'Mgmt/ProcessVersion': dbus.String('Unkown version, and running on Python 3.8.14', variant_level=1), 'Mgmt/Connection': dbus.String('MQTT PV service', variant_level=1), 'DeviceInstance': dbus.Int32(51, variant_level=1), 'ProductId': dbus.Int32(65535, variant_level=1), 'ProductName': dbus.String('MQTT PV', variant_level=1), 'CustomName': dbus.String('MQTT PV', variant_level=1), 'FirmwareVersion': dbus.String('0.1.4 (20231106)', variant_level=1), 'Connected': dbus.Int32(1, variant_level=1), 'Latency': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'ErrorCode': dbus.Int32(0, variant_level=1), 'Position': dbus.Int32(0, variant_level=1), 'StatusCode': dbus.Int32(0, variant_level=1), 'Ac/Power': dbus.Int32(0, variant_level=1), 'Ac/Current': dbus.Int32(0, variant_level=1), 'Ac/Voltage': dbus.Int32(0, variant_level=1), 'Ac/Energy/Forward': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Ac/L1/Power': dbus.Int32(0, variant_level=1), 'Ac/L1/Current': dbus.Int32(0, variant_level=1), 'Ac/L1/Voltage': dbus.In [... cut ...] '0', 'Ac/Power': '0W', 'Ac/Current': '0.0A', 'Ac/Voltage': '0.00V', 'Ac/Energy/Forward': '---', 'Ac/L1/Power': '0W', 'Ac/L1/Current': '0.0A', 'Ac/L1/Voltage': '0.00V', 'Ac/L1/Frequency': '---', 'Ac/L1/Energy/Forward': '---', 'Ac/MaxPower': '10000W', 'Ac/Position': '0', 'Ac/StatusCode': '0', 'UpdateIndex': '0'}
2023-11-07 14:05:37.378647500 DEBUG:root:PV: 222.0 W - 230.0 V - 1.0 A
2023-11-07 14:05:37.379922500 DEBUG:root:|- L1: 74.0 W - 233.0 V - 0.3 A

Output of MQTT Explorer looks good:
[image]

baetzst, 2023-11-07:
I got it.

In your config.sample.ini is no default frequency value.
If you don't publish frequency via MQTT you get the error:
`ERROR:root:Exception occurred: 'frequency'`
and only L1 is shown.
I added `frequency = 50` to config.ini and now it works.

Regards Stefan

closed 2023-11-07 by baetzst

mr-manuel, 2023-11-07:
Which version are you using? This should not happen with `v0.1.4`.

baetzst, 2023-11-07:
0.1.4
I copied today from github.

mr-manuel, 2023-11-07:
Ah, found it :-) thank you! I will correct that.