mr-manuel/venus-os_dbus-mqtt-pv #20: Wechselrichter nur an L2, dann teils unplausible Werte und Ansichten

mentos77, 2024-01-08:
Hallo Marcel, 
nutze seit paar Tagen dein Script, um die Daten vom Wechselrichter (erfasst mit OpenDTU) in Venus OS (auf Cerbo GX, v3.13) einzulesen und darzustellen. Mein Wechselrichter ist an L2 angeschlossen, demzufolge schicke ich auch nur Daten für L2, wie auf nachfolgenden „Dummydaten“ zu sehen:
[image]
In Venus OS (Detailansicht von diesem Gerät) werden jedoch identische Daten sowohl bei L1 als auch L2 angezeigt, bei L3 nur Bindestriche, was ja auch richtig ist. 
[image]

In der Übersicht
[image]

und Kachelansicht aber die Summe beider Phasen (also hier 2W).
[image]

Denke Ursache dafür ist in dbus-mqtt-pv.py die Zeilen 261-266. 
[image]

Kommentiere ich diese aus, werden nur noch Werte bei L2 und AC Gesamt angezeigt (jeweils 1W). Bei L1 stehen alle Werte auf 0, bei L3 sind weiterhin Bindestriche zu sehen. 
[image]

In den Übersichten wird die richtige Gesamtleistung von 1W angezeigt.
[image]
[image]

Das bei L1 weiterhin etwas angezeigt wird (0-Werte) liegt wohl an den Zeilen 392-396. 
[image]

Kommentiere ich diese ebenfalls aus, wird in der Detailansicht alles wie gewünscht angezeigt, sprich die richtigen Werte (1W bei L2 und AC Gesamt), bei L1 und L3 nur Bindestriche. 
[image]

In der Übersicht gibt es eine Anzeige für den PV-Wechselrichter, auch mit dem richtigen Wert von 1W.
[image]

Allerdings wird in der Kachelansicht gar kein PV-Wechselrichter mehr angezeigt. Die entsprechende Kachel zeigt das Victron Logo. 
[image]

Hast du eine Idee wie man das so hinbekommt, das überall das gewünschte angezeigt wird?
Danke und Gruß
Mentos77

mr-manuel, 2024-01-29:
Hallo @mentos77,

leider weiß ich nicht wen du mit Marcel meinst :-)

Ich habe den Treiber aktualisiert, kannst du er mit der letzten Version erneut versuchen?

mentos77, 2024-01-29:
Hallo Manuel,
sorry das ich den falschen Namen benutzt habe.
Habe den aktualisierten Treiber nun kurz getestet. Sieht soweit gut aus, Werte kommen nun richtig bei L2 an und auch beim Rest steht nix blödsinniges mehr. Mit anderen Phasen, auch in verschiedenen Kombinationen, funktioniert es auch.
Das Einzige, was noch nicht geht ist die Anzeige bei den Kacheln. Aber da mir diese Anzeige eh nicht gefällt ist es für mich kein weiteres Problem mehr.
Vielen Dank für das Update.
Gruß
Mentos77

closed 2024-01-29 by mentos77

mr-manuel, 2024-01-29:
Wichtig ist, dass es mit der GUIv2 funktioniert.