mr-manuel/venus-os_dbus-mqtt-pv #22: 0.1.6 does not update values

hanzoh, 2024-02-24:
After updating from 0.1.5 to 0.1.6 I noticed that after start of the plugin, it only updates once after startup and then keeps the same value even if the MQTT topic changes.

I reverted back to 0.1.5 and everything is fine again.

My topic's JSON looks like this:
```json
{"pv": {"power": 1829.0, "voltage": 235.03, "energy_forward": 34083.29 }}
```

This is the error I found in the log:
```
2024-02-24 07:57:48.572452500 *** starting dbus-mqtt-pv ***
2024-02-24 07:57:55.998191500 Traceback (most recent call last):
2024-02-24 07:57:55.998198500   File "/data/etc/dbus-mqtt-pv/dbus-mqtt-pv.py", line 263, in _update
2024-02-24 07:57:55.998202500     self._dbusservice['/Ac/L1/Power'] = round(pv_power, 2) if pv_power is not None else None
2024-02-24 07:57:55.998208500   File "/data/etc/dbus-mqtt-pv/ext/velib_python/vedbus.py", line 157, in __setitem__
2024-02-24 07:57:55.998212500     self._dbusobjects[path].local_set_value(newvalue)
2024-02-24 07:57:55.998216500 KeyError: '/Ac/L1/Power'
```

blaumsass, 2024-02-27:
I have update the json structur from mininimal to L1, L2, L3 and it run´s perfectly. iI think the problem is the version of Venus OS upper 3.19. Sorry for my english.

mr-manuel, 2024-02-27:
Not it's not related to the Venus OS version. It was due to a change in this code. I tested it and was not able to reproduce it after the latest fix.

closed 2024-02-27 by mr-manuel