mr-manuel/venus-os_dbus-mqtt-pv #24: VRM display with pv inverter MQTT

zigomatichub, 2024-03-21:
Hello,
I'm in the configuration (multiplus II) with the green part as solar injection and I added the mqtt to retrieve the information towards venus.
[image]
But when I look into VRM, it's not at the good place.
I tried all the positions of the config but this is always at same place or not visible.
[image]

Is it possible or it's not available from venus ?

mr-manuel, 2024-03-21:
Please check the Remote Console, if the placement is correct.

mr-manuel, 2024-03-22:
Ok, you are using GuiMods, therefore the correct position is not shown. Use position 1, even it it's not displaying correct.

For Enphase you are able to get real time data. Check [venus-os_dbus-enphase-envoy](https://github.com/mr-manuel/venus-os_dbus-enphase-envoy).

zigomatichub, 2024-03-22:
Thank you for your reply and suggestion

closed 2024-03-22 by zigomatichub