mr-manuel/venus-os_dbus-mqtt-pv #26: Odd readings

jeedee5, 2024-04-23:
Hi,

I installed the MQTT driver on my CerboGX, that is monitoring the PV's, but once it starts receiving data, the grid values on my victron starts acting up...

Following is my MQTT topic data
meter/pv {"pv": {"power": 6114.9, "L1": {"power": 0.0, "voltage": 231.9}, "L2": {"power": 3090.2, "voltage": 231.9}, "L3": {"power": 3024.7, "voltage": 231.9}}}

Without the MQTT meter:
[image]

Once the meter is reading data, I suddenly have a lot more energy going to the grid.
[image]

mr-manuel, 2024-04-24:
Your system is already wrong before. Why are the loads always 0?

mr-manuel, 2024-04-24:
Since this is a setup issue please check the Victron Community, which is the right place for your question.

closed 2024-04-24 by mr-manuel

jeedee5, 2024-04-24:
Fixed it... 

in the config.ini, i needed to change the position of the mqtt reader, so that is configured on the AC-output (while in reality it is on the AC-input)
; position (connected input/output) of the PV inverter
; 0 = AC input 1 (select this if you don't have a MultiPlus or Quattro)
; 1 = AC output
; 2 = AC input 2
position = 1

[image]