mr-manuel/venus-os_dbus-mqtt-pv #28: multible instances description nor clear

Autoschrauberix, 2024-05-05:
I try to get the data of 3 inverters into venus-os.
First is working perfect. Now I'm not shure how the MQTT data has to look like. For my understanding there have to be differences to the first instance. I didn't try it now but cat it be that the JSON structure has to start with "pv-2" for the second instance ?
Thank you very much for this implementation
Best regards

mr-manuel, 2024-05-05:
You just need a different topic for the second instance. The JSON structure is the same.

Autoschrauberix, 2024-05-05:
Thank you, that was it. I should have scoll down the config.ini , there is the topic for the instance I had missed.
I think it would be a improvement in the describtion to add at point 5 . Edit the topic for the new instance too.

Autoschrauberix, 2024-05-05:
Problem solved

closed 2024-05-05 by Autoschrauberix