mr-manuel/venus-os_dbus-mqtt-pv #30: Inverter Variables

MrJaroM, 2024-05-23:
Hello Mr. Manuel,
i have see in victron library this parameters:
[image]

i need this paramets too, in red for my inverter in your full doc. you say only:
{
    "pv": {
        "power": 0.0,
        "voltage": 0.0,
        "current": 0.0,
        "energy_forward": 0.0,           --> Total/Lifetime produced energy in kWh
        "L1": {
            "power": 0.0,
            "voltage": 0.0,
            "current": 0.0,
            "frequency": 0.0,
            "energy_forward": 0.0,       --> Total/Lifetime produced energy in kWh
        },
        "L2": {
            "power": 0.0,
            "voltage": 0.0,
            "current": 0.0,
            "frequency": 0.0,
            "energy_forward": 0.0,       --> Total/Lifetime produced energy in kWh
        },
        "L3": {
            "power": 0.0,
            "voltage": 0.0,
            "current": 0.0,
            "frequency": 0.0,
            "energy_forward": 0.0,       --> Total/Lifetime produced energy in kWh
        }
    }
}

Can I simply add these values in MQTT or does the interface have to be adapted to reflect them?

The next that I dont understand is "Yield" i receive from Inverter a value in Watt (every second) but i dont receive from inverter a value for Yield i can log all values and caluclate yield in kwh. 

My Opinion is I will emulate a EasySolar II.

Thanks for help:)

Is yield the same value from mppt tracker, that there get ?

mr-manuel, 2024-05-23:
See https://github.com/victronenergy/venus/wiki/dbus. Where did you get the Excel from?

`com.victronenergy.inverter` is not the same as `com.victronenergy.pvinverter`.

MrJaroM, 2024-05-24:
Thanks very much, for.
Can you help me by  a other question then. I will simulate a EasySolar II (PV inverter with one MPPT Tracker).
I have with your reps. two ideas 
1. add a solarcharger
2. and add a pvinverter

the best way where be simulate only one device...
Thanks