mr-manuel/venus-os_dbus-mqtt-pv #31: Question and animation issue

heggink, 2024-06-27:
Many thanks for creating this. I have it up and running pretty well.
I have a question and an observation:
Q: in the config file, there are 3 options for the AC side: 
; position (connected input/output) of the PV inverter
; 0 = AC input 1 (select this if you don't have a MultiPlus or Quattro)
; 1 = AC output
; 2 = AC input 2
position = 0
But the MP2 has one in and 2 outs, not the other way around. Thoughts?
Which leads me to the observation: if I use position=0 then there is no animation from the solar to the MP2. When I use position 1 then there IS animation but then vrm seems to list the AC load as including the PV power.
What is the right setting if my panels are connected to the AC IN side of the MP2 (ie directly tied to the grid)?

mr-manuel, 2024-06-27:
AC inout 2 is only usable on a Quattro. For a MultiPlus you can only use AC input 1 or AC output. The AC output 2 is only passthrough.
The correct one is AC input 1 (`position = 0`). Best is, when you check the Remote Console/GUI for the flow. The VRM does not always show correctly.

heggink, 2024-06-27:
Understood. Neither VRM nor the remote console show the animated flow. The values are shown but not a flow from the PV to the MP2

heggink, 2024-06-27:
[image]

mr-manuel, 2024-06-27:
You mean the electrons moving? They move only, if the power is above 30 W. Important is that the position and calculations are correct. The AC Loads are calculated and have to be correct.

heggink, 2024-06-27:
I'll send a screenshot tomorrow. So far, the electrons don't move at all.
Here's a copy of the mqtt msg I send:
{"pv": {"last_update": "2024-06-27 22:39:03", "power": 2, "energy_forward": 1200306.432, "L1": {"power": 2}, "L2": {"power": 0}, "L3": {"power": 0}}}

The power readings are taken from a power meter on the solar outpount where the sum is the calculated total. The energy_forward comes directly from the enphase (I cannot seem to get the power data per phase directly from enphase so attached a power meter). Updates are done every 30 seconds. As said, the data is all correct so veryhappy with that. I just miss the moving electrons :-).

mr-manuel, 2024-06-27:
The power of the PV in your sample is 2 W, way below 30 W.

You saw that repository? https://github.com/mr-manuel/venus-os_dbus-enphase-envoy

heggink, 2024-06-27:
I know. What I was trying to say is that it behaves the same above.
On your enphase plugin, I did see that. Unfortunately, I don't have an installer account and I have not found the time to figure out how to get it.

mr-manuel, 2024-06-28:
Try to ask in the Victron Community, since the driver works as expected and I do not really understand what your issue is.

heggink, 2024-06-28:
No worries, will do. In essence, I get all the data from your plugin. It's only the animation that is missing.

closed 2024-06-28 by mr-manuel