mr-manuel/venus-os_dbus-mqtt-pv #32: Wrong Calculation

breiti93, 2024-07-01:
Hello Manuel,

I have a similar issue like others before me but somehow I am not able to fix it.
Currently I am in an test mode:

MQTT GRID with real values
MQTT PV with real values
Multiplus Emulator

System is Loxone --> UDP to NodRED --> NodRED MQTT to VenusOS

But somehow my AC Loads are not calculated correct.
My AC Load is everytime exactly the load of my PV.

After installing the MQTT PV it works fine for 5-10 sec and after it changed to the wrong calculation.
I already tried different positions and also uninstall and install the converter again.

Do you have any idea what I can try?

[image]

Thanks in advance, 
Best regards,
Stefan

mr-manuel, 2024-07-02:
Without more details not. The position in your case should be `0`, since you have no real MultiPlus.

The load in this case should be 1700 - 1114 = 586 W right?

Did you select under Settings -> System setup -> AC input 1 and AC input 2? AC input 1 should be `Grid` and AC input 2 should be `Not available`.

The page under Settings -> ESS is empty, right?

mr-manuel, 2024-07-02:
You could also try to use a different version of the MultiPlus emulator: https://github.com/mr-manuel/venus-os_dbus-multiplus-emulator/releases/tag/v0.0.3

mr-manuel, 2024-07-02:
Btw, why are you using the MultiPlus emulator? In your case it should not be needed.

breiti93, 2024-07-02:
ahh damn... I have read in some Post that its necessary for recognising the PV and calculating everything correct.
But than i was exaclty the oposite.
Without the emulator in looks better

mhin76, 2024-07-02:
@breiti93 I have same issue. I have OpenDTU as PV service and a mqtt battery service setup. to have the grid calculated I had to install the MultiPlus emulator. but that does not work out if I chose any phase within the emulator. If I set the phase to "L4" which does not exist, all looks fine, but battery power is not considered. in your case that would however solve the issue already. Bc with L4 a lot of calculation in the MultiPlus emulator are skipped. I am debugging it to find out, what is going wrong.

What does manuel mean with position 0?

mr-manuel, 2024-07-02:
@breiti93 if I remember correctly, you need the MultiPlus emulator only, if you have AC + DC values. Like, if you are transfering power from AC to DC or from DC to AC.

@mhin76 
https://github.com/mr-manuel/venus-os_dbus-mqtt-pv/blob/a837a8593cd06189fe1b3cfcea127eda7ddd8a12/dbus-mqtt-pv/config.sample.ini#L37-L41

mhin76, 2024-07-02:
the MultiPlus emulator is required as soon as you have a PV inverter connected and want to see the Grid amount. Issue without emulator is described here: https://community.victronenergy.com/questions/205392/no-ac-input-on-vrm.html

I am using the openDTU for inverter. need to check on that setting as well :)

breiti93, 2024-07-02:
I think the behavior changed in the lates Version of VenusOS.
I uninstalled the Emulator and now it looks great:

[image]

mr-manuel, 2024-07-02:
I think also that there was modified something.

mhin76, 2024-07-02:
> I think the behavior changed in the lates Version of VenusOS. I uninstalled the Emulator and now it looks great: [...]

yes, that worked for me before. but do you see the sum of the AC in on VRM? that was the issue I had and only the emulator solved it. I have also set the PV inverter with AC position = 0.