mr-manuel/venus-os_dbus-mqtt-pv #36: Only L1 after updating from 0.1.2 to 0.1.7

marcelkrois, 2024-07-22:
Hi manuel, hi all,

after updating from 0.1.2. to 0.1.7 it was only actualization power and energy forward and L1 (voltage, current and power). L2 and L3 wasn't updating in Victron Dashboard.
The MQTT Massage was not change:
This is my MQTT:
`{
  "pv": {
    "power": 6912,
    "energy_forward": 31171,
    "L1": {
      "power": 2105,
      "voltage": 245,
      "current": 8.856
    },
    "L2": {
      "power": 2414,
      "voltage": 242,
      "current": 10.181
    },
    "L3": {
      "power": 2393,
      "current": 10.501,
      "voltage": 241
    }
  }
}`

[image]

Is there something wrong?

To fix it i have overwrite line 146 to 179 with the lines from 0.1.2 in dbus-mqtt-pv.py.
After change, all phases are updated. 

in 0.1.2 it was:
`if 'pv' in jsonpayload:
                    if type(jsonpayload['pv']) == dict and 'power' in jsonpayload['pv']:
                        pv_power = float(jsonpayload['pv']['power'])
                        pv_current = float(jsonpayload['pv']['current']) if 'current' in jsonpayload['pv'] else pv_power/float(config['DEFAULT']['voltage'])
                        pv_voltage = float(jsonpayload['pv']['voltage']) if 'voltage' in jsonpayload['pv'] else float(config['DEFAULT']['voltage'])
                        pv_forward = float(jsonpayload['pv']['energy_forward']) if 'energy_forward' in jsonpayload['pv'] else 0

                        # check if L1 and L1 -> power exists
                        if 'L1' in jsonpayload['pv'] and 'power' in jsonpayload['pv']['L1']:
                            pv_L1_power = float(jsonpayload['pv']['L1']['power'])
                            pv_L1_current = float(jsonpayload['pv']['L1']['current']) if 'current' in jsonpayload['pv']['L1'] else pv_L1_power/float(config['DEFAULT']['voltage'])
                            pv_L1_voltage = float(jsonpayload['pv']['L1']['voltage']) if 'voltage' in jsonpayload['pv']['L1'] else float(config['DEFAULT']['voltage'])
                            pv_L1_forward = float(jsonpayload['pv']['L1']['energy_forward']) if 'energy_forward' in jsonpayload['pv']['L1'] else 0

                        # check if L2 and L2 -> power exists
                        if 'L2' in jsonpayload['pv'] and 'power' in jsonpayload['pv']['L2']:
                            pv_L2_power = float(jsonpayload['pv']['L2']['power'])
                            pv_L2_current = float(jsonpayload['pv']['L2']['current']) if 'current' in jsonpayload['pv']['L2'] else pv_L2_power/float(config['DEFAULT']['voltage'])
                            pv_L2_voltage = float(jsonpayload['pv']['L2']['voltage']) if 'voltage' in jsonpayload['pv']['L2'] else float(config['DEFAULT']['voltage'])
                            pv_L2_forward = float(jsonpayload['pv']['L2']['energy_forward']) if 'energy_forward' in jsonpayload['pv']['L2'] else 0

                        # check if L3 and L3 -> power exists
                        if 'L3' in jsonpayload['pv'] and 'power' in jsonpayload['pv']['L3']:
                            pv_L3_power = float(jsonpayload['pv']['L3']['power'])
                            pv_L3_current = float(jsonpayload['pv']['L3']['current']) if 'current' in jsonpayload['pv']['L3'] else pv_L3_power/float(config['DEFAULT']['voltage'])
                            pv_L3_voltage = float(jsonpayload['pv']['L3']['voltage']) if 'voltage' in jsonpayload['pv']['L3'] else float(config['DEFAULT']['voltage'])
                            pv_L3_forward = float(jsonpayload['pv']['L3']['energy_forward']) if 'energy_forward' in jsonpayload['pv']['L3'] else 0`

Regards 
Marcel

dixi83, 2024-08-21:
Hi Marcel,
I had the same issue, 
`tail -n 100 -f /data/log/dbus-mqtt-pv/current | tai64nlocal`
showed me:
```
2024-08-21 14:08:59.106180500 ERROR:root:Exception occurred: KeyError('voltage') of type <class 'KeyError'> in /data/etc/dbus-mqtt-pv/dbus-mqtt-pv.py line #166
```
What pointed me to that the script was missing the DEFAULT[voltage] was missing...

So, you probably commented out these lines in the config.ini: 
```ini
; used when no voltage is received
voltage = 230

; used when no frequency is received
frequency = 50
```
Hope, this helped!

mr-manuel, 2024-10-08:
Probably the `fequency` setting is missing in your `config.ini`.

https://github.com/mr-manuel/venus-os_dbus-mqtt-pv/blob/e124e19098909eea83047d566f22f3a7ab2e58ea/dbus-mqtt-pv/config.sample.ini#L27-L31

closed 2024-10-08 by mr-manuel