mr-manuel/venus-os_dbus-mqtt-pv #38: Connecting a Huawei SUN2000

krulkip, 2024-08-19:
I have above inverter. It is connected to home assistant and i can see the Power values there.
Now i try to publish them on MQTT and then read them in the VenusOS.
I use below script and can see the message in MQTT explorer. like so 
PVSun2000 = {"pv": {"power": 2161,"L3": {"power": 2161}}}

service: mqtt.publish
data:
  qos: "0"
  retain: true
  topic: PVSun2000
  payload: "{\"pv\": {\"power\": {{states('sensor.zonneplan_last_measured_value')}},\"L3\": {\"power\": {{states('sensor.zonneplan_last_measured_value')}}}}}"

i installed https://github.com/mr-manuel/venus-os_dbus-mqtt-pv as described but do not see it on console
i have another PV inverter connected through shelly
Is that possible or what am i doing wrong

mr-manuel, 2024-08-20:
Did you troubleshoot the driver? What does the log say and how does your config.ini look like?

krulkip, 2024-08-20:
Thank you for your reply. Troubleshooting the driver ... current file in /data/log showed the frequency of sending mqtt messages was not enough and the timeout reset the driver. Corrected that and now all is working

closed 2024-08-20 by mr-manuel