mr-manuel/venus-os_dbus-mqtt-pv #42: on_message doesn't get called

randomname32, 2024-11-24:
Hi, I have a problem that the "on_message" function doesn't get called.
The logfile states that the connection to the broker succeeds (broker_address = 192.168.188.58) 
I am subscribing to the topic "balkonsolar".
Using an mqtt-explorer, I can see that my device (a shelly 1pm) publishes mqtt messages to that topic (see screenshot). I know that the payload isn't correct (yet), but that is not the reason for my issue, because the payload is only evaluated inside the on_message function.

As a test, I changed the topic in the config to "grid", this executed the on_message function.

I am using dbus-mqtt-grid in parallel, which works fine.

Is maybe running 2 dbus-mqtt-X instances in parallel a problem?

[image]

mr-manuel, 2024-11-25:
This is not how MQTT works. I would recommend you, that you watch some videos about how MQTT really works.

A parent topic cannot have any data, therefore it will never update. You only can use topics that contain some "direct" data and not any subtopics.

closed 2024-11-25 by mr-manuel

randomname32, 2024-11-25:
Thanks for your quick reply and sorry for my stupidity. I tried it with the correct prefix and now it works.