mr-manuel/venus-os_dbus-mqtt-pv #46: data not displayed

grydo, 2025-01-01:
Hi, data in gui not displayed 
[image]
[image]

my config.ini

; CONFIG FILE
; GitHub reporitory: https://github.com/mr-manuel/venus-os_dbus-mqtt-pv
; remove semicolon ; to enable desired setting

[DEFAULT]
; Set logging level
; ERROR = shows errors only
; WARNING = shows ERROR and warnings
; INFO = shows WARNING and running functions
; DEBUG = shows INFO and data/values
; default: WARNING
logging = INFO

; Device name
; default: MQTT PV
device_name = MQTT PV

; Device VRM instance
; default: 51
device_instance = 51

; Specify after how many seconds the driver should exit (disconnect), if no new MQTT message was received
; default: 60
; value to disable timeout: 0
timeout = 60

; used when no voltage is received
voltage = 230

; used when no frequency is received
frequency = 50

[PV]
; Max rated power (in Watts) of the inverter
max = 3500
; position (connected input/output) of the PV inverter
; 0 = AC input 1 (select this if you don't have a MultiPlus or Quattro)
; 1 = AC output
; 2 = AC input 2
position = 0

[MQTT]
; IP addess or FQDN from MQTT server
broker_address = 192.168.10.133

; Port of the MQTT server
; default plaintext: 1883
; default TLS port: 8883
broker_port = 1883

; Enables TLS
; 0 = Disabled
; 1 = Enabled
;tls_enabled = 1

; Absolute path to the Certificate Authority certificate file that is to be treated as trusted by this client
;tls_path_to_ca = /data/keys/mosquitto.crt

; Disables verification of the server hostname in the server certificate
; 0 = Disabled
; 1 = Enabled
;tls_insecure = 1

; Username used for connection
username = mqtt1

; Password used for connection
password = mqtt1

; Topic where the pv data as JSON string is published
; minimum required JSON payload: {"pv": { "power": 0.0 } }
topic = enphase/envoy-s/meters

my json home assistant

alias: mqtt publish sensor pv power
description: ""
triggers:
  - entity_id: sensor.ac_output_active_power
    trigger: state
conditions: []
actions:
  - data_template:
      payload: |
        {
          "pv": {
            "power": {{ (states('sensor.ac_output_active_power') | float(0)) }},
            "L1": {
                "power": {{ (states('sensor.ac_output_active_power') | float(0)) }}
            }
          }
        }
      topic: enphase/envoy-s/meters
    action: mqtt.publish

mr-manuel, 2025-01-01:
Not work if only mqtt pv. You need Victron.

grydo, 2025-01-01:
Work only inverter Victron not other inverter?

mr-manuel, 2025-01-01:
Why, do you want to use Venus OS, if you else seem to have no other Victron hardware?

Please describe your complete setup, what you are using as a GX device, and what you try to achieve else it's hard to help you.

grydo, 2025-01-01:
Venus os installed in raspberry pi, 2 mppt victron, 1 battery can installed, my inverter is Voltronic,  I would like to display the AC load in VRM.

mr-manuel, 2025-01-01:
Any why is nothing you mentioned visible in Venus OS?

So this is an off grid installation? Why are you trying to add AC PV if you have DC PV?

mr-manuel, 2025-01-01:
You could try, if dbus-multiplus-emulator works in the setup you posted. For sure it won't work with the first approach you had.

Anyway there is no bug in this driver, so I close this issue. If you have further questions ask the Victron Community.

closed 2025-01-01 by mr-manuel