mr-manuel/venus-os_dbus-mqtt-pv #47: Not persisting across reboots.

sticilface, 2025-01-06:
I'm using the rPi image of Venus OS:  latest version ([venus-image-large-raspberrypi2-20241120085113-v3.52.rootfs.wic.gz](https://updates.victronenergy.com/feeds/venus/release/images/raspberrypi2/venus-image-large-raspberrypi2-20241120085113-v3.52.rootfs.wic.gz)

I've followed the instructions precisely... copying the links. 

``
wget -O /tmp/download_dbus-mqtt-pv.sh https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-mqtt-pv/master/download.sh
bash /tmp/download_dbus-mqtt-pv.sh
nano /data/etc/dbus-mqtt-pv/config.ini - updated settings
bash /data/etc/dbus-mqtt-pv/install.sh
```

<img width="1086" alt="image" src="https://github.com/user-attachments/assets/df802f88-8086-4267-8b18-8417d2afa803" />
Solar appears fine.  

Only when I reboot it is not there.  The same occurs with you virtual multiplus.  There are no other devices or services installed or running over the base image.  On reboot there is no `dbus-mqtt-pv` in services..  it appears if i rerun the install script. 

Any pointers?

sticilface, 2025-01-06:
Solved my own issue.   my rc.local had an exit 0 as the last line before installation, so the lines were appended incorrectly.  

[image]

Just in case anyone else has the same issue.

mr-manuel, 2025-01-06:
Why is there an exit in this script? Did you do it yourself?

closed 2025-01-06 by mr-manuel

sticilface, 2025-01-06:
I've got a shelly EM and a solis inverter with modbusTCP...  I've been trying to get the solar data to appear in the Venus using a variety of of techniques.  One was a custom modbus client which has its own install script, and i copied and pasted from their howto.  I've kept the issue and put it here incase other people have the same.  I've now removed it and its works well.  Thank you