mr-manuel/venus-os_dbus-mqtt-pv #49: AC Loads incorrect

sticilface, 2025-01-25:
Thank you for a great bit of software.  It is working really well... except my AC loads are incorrect and reading zero.  I've read the variety of issues relating to this and my configuration should be correct.  

I have an AC PV inverter on the AC input of the Multiplus.  position = 0 in the config.  It appears on the left of the multiplus.  Within the cerboGX I have configured loads on input and output of the multiplus.  

[image]

[image]

[image]

any further suggestions?

sticilface, 2025-01-25:
with no solar the accounting is correct.

[image]

mr-manuel, 2025-01-26:
Do you have a physical MultiPlus? If not use the dbus-multiplus-emulator in my repository.

sticilface, 2025-01-26:
It's a physical multiplus 10kVa in a fully working system. 

On the VRM it's correctly located on the AC input as well. 

Could the inverter status be the issue? Ie not sure the running field is there?

mr-manuel, 2025-01-26:
No the inverter status should be no problem. In this case it's probably better if you ask the Victron community, since it's very likely a system config issue.

closed 2025-01-28 by mr-manuel

sticilface, 2025-01-28:
So the having learnt a lot over the last few days.. my believe is that the multiplus only has one current sensor and if you are using that for grid metering (by selecting inverter/multiplus) as the grid meter in ESS (ie not external meter).. then it looses the ability to tell you where your loads are.  

I just purchased the victron  VM-3P75CT plugged it in... and boom it's working as expected.  

[image]

Thought i'd follow this up for reference incase anyone else finds the same.