mr-manuel/venus-os_dbus-mqtt-pv #50: Venus OS Solar Inverters showing but no data

lukaszlk, 2025-01-31:
Hi,

Im using Homeassistant with Mosquito Broker. My VenusOs Multiplus is connecting and a category of Solar Inverters is present, and in Devices MQTT PV is present as well, but it's not displaying any data from the sensors I've provided. Any idea where I'm making a mistake? Below is the automatization I'm using to publish my 3 inverters and a sensor which sums up those 3 values. In anybody could help me with that I would be grateful.

alias: mqtt publish sensor pv power
description: ""
triggers:
  - entity_id: sensor.total_pv_power_dynamic
    trigger: state
conditions: []
actions:
  - data:
      topic: homeassistant/energy/pv
      payload: |
        {
          "pv": {
            "power": {{ states('sensor.total_pv_power_dynamic') | float(0) }},
            "L1": {
              "power": {{ states('sensor.lukasz_pv_power') | float(0) }}
            },
            "L2": {
              "power": {{ states('sensor.sensor.micro_inverter_garage_power') | float(0) }}
            },
            "L3": {
              "power": {{ states('sensor.micro_inverter_garage_power_2') | float(0) }}
            }
          }
        }
    action: mqtt.publish

mr-manuel, 2025-01-31:
https://github.com/mr-manuel/venus-os_dbus-mqtt-pv?tab=readme-ov-file#debugging

lukaszlk, 2025-01-31:
2025-01-31 07:52:04.071551500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 07:52:04.072221500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 07:52:04.260240500 *** starting dbus-mqtt-pv ***
2025-01-31 07:53:05.707828500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 07:53:05.708552500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 07:53:05.906830500 *** starting dbus-mqtt-pv ***
2025-01-31 07:54:07.386891500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 07:54:07.387573500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 07:54:07.578026500 *** starting dbus-mqtt-pv ***
2025-01-31 07:56:30.361663500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 07:56:30.556175500 *** starting dbus-mqtt-pv ***
2025-01-31 07:57:32.046423500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 07:57:32.047095500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 07:57:32.246562500 *** starting dbus-mqtt-pv ***

lukaszlk, 2025-01-31:
Closed, the issue was different measurements, both in kWh and in W, after setting all sensors to W it works.