mr-manuel/venus-os_dbus-mqtt-pv #51: Connection with MQTT broker issue

lukaszlk, 2025-02-04:
Hi,

I have an issue with what seems as an issue with connection with mosquito broker in Homeassistant. Looks like after a couple of timeouts when inverters go offline and they are not publishing any data no more, the connection is initialized but gets stuck and is not reporting when inverters go online in the morning. Anyone had this issue? Thanks

 

 

 2025-02-03 22:08:09: New client connected from 192.168.0.253:39617 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:09:14: Client MqttPv_51 closed its connection.
2025-02-03 22:09:16: New client connected from 192.168.0.253:58293 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:10:21: Client MqttPv_51 closed its connection.
2025-02-03 22:10:22: New client connected from 192.168.0.253:37799 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:11:28: Client MqttPv_51 closed its connection.
2025-02-03 22:11:29: New client connected from 192.168.0.253:51489 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:12:34: Client MqttPv_51 closed its connection.
2025-02-03 22:12:36: New client connected from 192.168.0.253:60741 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:13:41: Client MqttPv_51 closed its connection.
2025-02-03 22:13:42: New client connected from 192.168.0.253:41875 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:14:48: Client MqttPv_51 closed its connection.
2025-02-03 22:14:49: New client connected from 192.168.0.253:39263 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:15:54: Client MqttPv_51 closed its connection.
2025-02-03 22:15:56: New client connected from 192.168.0.253:34193 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:17:01: Client MqttPv_51 closed its connection.
2025-02-03 22:17:02: New client connected from 192.168.0.253:48209 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:18:08: Client MqttPv_51 closed its connection.
2025-02-03 22:18:09: New client connected from 192.168.0.253:38829 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:19:14: Client MqttPv_51 closed its connection.
2025-02-03 22:19:16: New client connected from 192.168.0.253:39803 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:20:21: Client MqttPv_51 closed its connection.
2025-02-03 22:20:23: New client connected from 192.168.0.253:57271 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:21:28: Client MqttPv_51 closed its connection.
2025-02-03 22:21:29: New client connected from 192.168.0.253:58699 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:22:34: Client MqttPv_51 closed its connection.
2025-02-03 22:22:36: New client connected from 192.168.0.253:47597 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:23:41: Client MqttPv_51 closed its connection.
2025-02-03 22:23:43: New client connected from 192.168.0.253:34035 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:24:48: Client MqttPv_51 closed its connection.
2025-02-03 22:24:49: New client connected from 192.168.0.253:47283 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:25:55: Client MqttPv_51 closed its connection.
2025-02-03 22:25:56: New client connected from 192.168.0.253:46507 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:27:01: Client MqttPv_51 closed its connection.
2025-02-03 22:27:03: New client connected from 192.168.0.253:47927 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:28:08: Client MqttPv_51 closed its connection.
2025-02-03 22:28:09: New client connected from 192.168.0.253:44483 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:29:15: Client MqttPv_51 closed its connection.
2025-02-03 22:29:16: New client connected from 192.168.0.253:40179 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:30:21: Client MqttPv_51 closed its connection.
2025-02-03 22:30:23: New client connected from 192.168.0.253:43551 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:31:28: Client MqttPv_51 closed its connection.
2025-02-03 22:31:29: New client connected from 192.168.0.253:42651 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:32:35: Client MqttPv_51 closed its connection.
2025-02-03 22:32:36: New client connected from 192.168.0.253:47737 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:33:41: Client MqttPv_51 closed its connection.
2025-02-03 22:33:43: New client connected from 192.168.0.253:38431 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:34:48: Client MqttPv_51 closed its connection.
2025-02-03 22:34:49: New client connected from 192.168.0.253:33541 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:35:55: Client MqttPv_51 closed its connection.
2025-02-03 22:35:56: New client connected from 192.168.0.253:54543 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:37:01: Client MqttPv_51 closed its connection.
2025-02-03 22:37:03: New client connected from 192.168.0.253:41521 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:38:08: Client MqttPv_51 closed its connection.
2025-02-03 22:38:09: New client connected from 192.168.0.253:50531 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:39:15: Client MqttPv_51 closed its connection.
2025-02-03 22:39:16: New client connected from 192.168.0.253:43809 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:40:21: Client MqttPv_51 closed its connection.
2025-02-03 22:40:23: New client connected from 192.168.0.253:34025 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:41:28: Client MqttPv_51 closed its connection.
2025-02-03 22:41:29: New client connected from 192.168.0.253:45573 as MqttPv_51 (p2, c1, k60, u'ha').
2025-02-03 22:42:35: Client MqttPv_51 closed its connection.
[... 82 lines cut ...]
2025-01-31 16:32:32.800974500 *** starting dbus-mqtt-pv ***
2025-01-31 16:33:34.550232500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 16:33:34.550924500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 16:33:34.783228500 *** starting dbus-mqtt-pv ***
2025-01-31 16:34:36.662177500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 16:34:36.662895500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 16:34:36.874697500 *** starting dbus-mqtt-pv ***
2025-01-31 16:35:38.625145500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 16:35:38.626017500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 16:35:38.834174500 *** starting dbus-mqtt-pv ***
2025-01-31 16:36:40.609653500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 16:36:40.610478500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 16:36:40.843967500 *** starting dbus-mqtt-pv ***
2025-01-31 16:37:42.604808500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 16:37:42.605518500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 16:37:42.821871500 *** starting dbus-mqtt-pv ***
2025-01-31 16:38:44.612458500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 16:38:44.613258500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 16:38:44.823191500 *** starting dbus-mqtt-pv ***
2025-01-31 16:39:46.680078500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 16:39:46.680868500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 16:39:46.893876500 *** starting dbus-mqtt-pv ***
2025-01-31 16:40:48.699954500 WARNING:root:Waiting since 60 seconds for receiving first data...
2025-01-31 16:40:48.700634500 ERROR:root:Driver stopped. Timeout of 60 seconds exceeded, since no new MQTT message was received in this time.
2025-01-31 16:40:48.911454500 *** starting dbus-mqtt-pv ***

mr-manuel, 2025-02-06:
Please post your config files.

lukaszlk, 2025-02-07:
; CONFIG FILE
; GitHub reporitory: https://github.com/mr-manuel/venus-os_dbus-mqtt-pv
; remove semicolon ; to enable desired setting
[DEFAULT]
; Set logging level
; ERROR = shows errors only
; WARNING = shows ERROR and warnings
; INFO = shows WARNING and running functions
; DEBUG = shows INFO and data/values
; default: WARNING
logging = DEBUG

; Device name
; default: MQTT PV
device_name = MQTT PV

; Device VRM instance
; default: 51
device_instance = 51

; Specify after how many seconds the driver should exit (disconnect), if no new                                                                                                                            >
; default: 60
; value to disable timeout: 0
timeout = 61

; used when no voltage is received
voltage = 230

; used when no frequency is received
frequency = 50

[PV]
; Max rated power (in Watts) of the inverter
max = 3500
; position (connected input/output) of the PV inverter
; 0 = AC input 1 (select this if you don't have a MultiPlus or Quattro)
; 1 = AC output
; 2 = AC input 2
position = 2

[MQTT]
; IP addess or FQDN from MQTT server
broker_address = 192.168.0.10x

; Port of the MQTT server
; default plaintext: 1883
; default TLS port: 8883
broker_port = 1883

; Enables TLS
; 0 = Disabled
; 1 = Enabled
;tls_enabled = 1
; Absolute path to the Certificate Authority certificate file that is to be tre                                                                                                                            >
;tls_path_to_ca = /data/keys/mosquitto.crt

; Disables verification of the server hostname in the server certificate
; 0 = Disabled
; 1 = Enabled
;tls_insecure = 1

; Username used for connection
username = ha

; Password used for connection
password = password

; Topic where the pv data as JSON string is published
; minimum required JSON payload: {"pv": { "power": 0.0 } }
topic = homeassistant/energy/pv

The issue is that once I get it to work, it works fine until, when there are no more values published by Homeassistant, I get a Mosquito Log - 2025-02-06 16:10:47: Client MqttPv_51 closed its connection and the service does not start up again the next day. On top of that the logs via tail -n 100 -F /data/log/dbus-mqtt-pv/current | tai64nlocal are stuck in January and the last time I got the service working was yesterday.

lukaszlk, 2025-02-08:
Will try that, so far no disconnect logs in Mosquito so I will see tomorrow

lukaszlk, 2025-02-11:
Since the connection is not lost anymore and the service shows the last positive value of around 2W untill the next day it Works nów and doesn’t disconnect, thanks.

closed 2025-02-11 by lukaszlk

grydo, 2026-03-12:
@lukaszlk Could you tell me how you publish the homeassistant values ​​on venus os?