mr-manuel/venus-os_dbus-mqtt-pv #57: Not show total sun

broekieman, 2025-03-30:
Is it not possible to show total sun? 

[image]

mr-manuel, 2025-03-31:
You have to provide energy forward for that.

broekieman, 2025-03-31:
Is this a setting, please explain.

mr-manuel, 2025-03-31:
Read the README and check the [full JSON structure](https://github.com/mr-manuel/venus-os_dbus-mqtt-pv#json-structure).

broekieman, 2025-03-31:
We solved this code for homeassistant together(missed a komna) I have pv data in my screenshot but Totaal stays zero.

RandomVictronUser, 2025-04-09:
Just to let you know. Your issue is no problem here.

broekieman, 2025-04-09:
> Just to let you know. Your issue is no problem here.

You mean, you have also enphase standaar and the script and you get field total filled?

mr-manuel, 2025-04-09:
Why are you not using directly the Enphase driver in my repository?

RandomVictronUser, 2025-04-09:
> > Just to let you know. Your issue is no problem here.
> 
> You mean, you have also enphase standaar and the script and you get field total filled?

Yup. Enphase Envoy, Relay and IQ8 micro’s.

broekieman, 2025-04-09:
> > > Just to let you know. Your issue is no problem here.
> > 
> > 
> > You mean, you have also enphase standaar and the script and you get field total filled?
> 
> Yup. Enphase Envoy, Relay and IQ8 micro’s.

Envoy standard or envoy metered?

RandomVictronUser, 2025-04-09:
> Why are you not using directly the Enphase driver in my repository?

For me that is due to the fact that I use the Envoy Gateway S. Not a metered Envoy gateway. Next to a couple of IQ8’s that are still coupled via HomeWizard Energy Plugs.

RandomVictronUser, 2025-04-09:
> > > > Just to let you know. Your issue is no problem here.
> > > 
> > > 
> > > You mean, you have also enphase standaar and the script and you get field total filled?
> > 
> > 
> > Yup. Enphase Envoy, Relay and IQ8 micro’s.
> 
> Envoy standard or envoy metered?

Enphase Gateway S (new name). Not a metered one.

RandomVictronUser, 2025-04-09:
> Please show your photo like mine.

[image]

broekieman, 2025-04-09:
So you do not have ct clamps?

RandomVictronUser, 2025-04-09:
Not for the Enphase Gateway Standard in a shed. Only for the new Victron Energy meter. Which works with CT’s as you may know. 

But how is your JSON configured? My guess is that there is some omission or error. Have you checked cat /data/log/dbus-mqtt-pv/current and view -> debug messaged?

broekieman, 2025-04-09:
Do you mean json in home assstant?
Please share yours.

broekieman, 2025-04-10:
I do not have the fles current and view.

RandomVictronUser, 2025-04-10:
Sorry. I don't use Home Assistant (as in not yet) and my data won't help you either.

broekieman, 2025-04-10:
Still you are using this plugin and receive data in the venus.

broekieman, 2025-04-11:
Please tell me which envoy sensors i need for 1 phase installation.

payload: |
        {
          "pv": {
            "power": {{ (states('sensor.YOUR_PV_POWER_ENTITY') | float(0)) }},
            "L1": {
                "power": {{ (states('sensor.YOUR_PV_L1_POWER_ENTITY') | float(0)) }}

mr-manuel, 2025-04-11:
I can't know how the sensor in your system is named. You need to add the lifetime produced energy as `energy_forward`.

broekieman, 2025-04-11:
I was forgotten the username/password.
It's working now.
I have 1 phase enphase system on L1.
What do i need to get the total filled?

[image]

Is my j.son ok?

{
  "pv": {
    "power": {{ (states('sensor.envoy_122248025029_current_power_production') | float(0)) }},
    "L1": {
        "power": {{ (states('sensor.envoy_122248025029_current_power_production') | float(0)) }}
    }
  }
}

RandomVictronUser, 2025-04-11:
You need to add the  “energy_forward:” total_energy property, where the latter represents the total production data of the envoy. I use total_energy = Math.trunc(energy);

broekieman, 2025-04-11:
Is this correct?

``
params:
  domain: mqtt
  service: publish
  service_data:
    payload: |-
      {
        "pv": {
          "power": 0.0,
          "L1": {
              "power": 0.0
        "Energy forward": {
          "power": 11051784.0  
          }
        }
      }
    topic: homeassistant/energy/pv
``

mr-manuel, 2025-04-12:
What are you using to get the PV data into home assistant? Would be better, if you ask there for help.

RandomVictronUser, 2025-04-12:
> Yes i have D8 firmware but i do not understand how to add the énergy_forward:'total_energy property. Can you share an example?

I think that you need to add a similar line like you have for wattage, but then replace the current_power_production with Lifetime_energy_production

Again. I do not use HomeAssistant so this is my best guess. Sorry if that doesn’t work. Share JSON when it fails or works. Thanks.

broekieman, 2025-04-12:
I read following in py file:

 if "energy_forward" in jsonpayload["pv"]["L1"]:
                                    pv_L1_forward = float(jsonpayload["pv"]["L1"]["energy_forward"])

How does this look like in j.son when only having L1 configured?

mr-manuel, 2025-04-12:
I don't know what you don't get, but here is everything listed again:

Needed JSON payload as written in the README that the driver needs to display correctly in VRM:

```json
{
    "pv": {
        "power": 0.0,
        "energy_forward": 0.0
    }
}
```

OR

```json
{
    "pv": {
        "power": 0.0,
        "energy_forward": 0.0,
        "L1": {
            "power": 0.0,
            "energy_forward": 0.0
        }
    }
}
```

Possible Home Assistant config (out of scope of this repository):

```yml
alias: mqtt publish sensor pv power
description: ""
trigger:
  - platform: state
    entity_id: sensor.YOUR_PV_POWER_ENTITY
condition: []
action:
  - service: mqtt.publish
    data_template:
      payload: |
        {
          "pv": {
            "power": {{ (states('sensor.YOUR_PV_POWER_ENTITY') | float(0)) }},
            "energy_forward": {{ (states('sensor.YOUR_PV_LIFETIME_PRODUCTION_IN_KWH') | float(0)) }},
            "L1": {
                "power": {{ (states('sensor.YOUR_PV_L1_POWER_ENTITY') | float(0)) }}
                "energy_forward": {{ (states('sensor.YOUR_PV_LIFETIME_PRODUCTION_IN_KWH') | float(0)) }}
            }
          }
        }
      topic: homeassistant/energy/pv
```

OR

```yml
alias: mqtt publish sensor pv power
description: ""
trigger:
  - platform: state
    entity_id: sensor.YOUR_PV_POWER_ENTITY
condition: []
action:
  - service: mqtt.publish
    data_template:
      payload: |
        {
          "pv": {
            "power": {{ (states('sensor.YOUR_PV_POWER_ENTITY') | float(0)) }},
            "energy_forward": {{ (states('sensor.YOUR_PV_LIFETIME_PRODUCTION_IN_KWH') | float(0)) }}
          }
        }
      topic: homeassistant/energy/pv
```

Maybe this could also help you: https://github.com/Matthew1471/Enphase-API

How you get the data from your device and how you build it in Home Assistant is out of the scope of this repository.

closed 2025-04-12 by mr-manuel

broekieman, 2025-04-14:
Any idea why my value is so high? 

[image]

broekieman, 2025-04-16:
[image]

You should use: today total production, instead off lifetime energy.