mr-manuel/venus-os_dbus-mqtt-pv #64: Inverter not showing in Remote Console and VRM

Ravenor, 2025-07-04:
Hi Manuel,

i reinstalled the venus-os_dbus-mqtt-pv on my Cerbo GX because there was an FW-Update of the Cerbo and after installing the FW v3.62 my PV Inverter was gone.
After reinstalling the driver, i could not get it to work any more.  
This is my MQTT-Topic

[image]

config.ini

[config.txt](https://github.com/user-attachments/files/21066072/config.txt)

I don't know whats wrong.
My you have an Idea.
Sorry for my bad english, i'm german.
Thank you very much.

Andy

mr-manuel, 2025-07-05:
You need one topic with all the values, see the README.

Ravenor, 2025-07-05:
Thank you for your reply. Now i have to figure out how to get al these payloads(L1,L2,L3,etc.), in node red, into one topic .
I build the topic above for the last version i used and it worked. I don't know how, but it did. 
I hope i find some videos, explainig how it has to work, bcause i am not a programmer like you are.
Thasnk you.

Ravenor, 2025-07-06:
It is working partialy. But it is only reading power.

[image]
I don't know how to post all Values at once, so i post them one after the other. 
Not working is: voltage is only used under L1 and it's from config.ini, energy_forward nowhere, power is used under L1 and above, current is the same L1 and above.
I read the readme, but it don't helps.
If my Problem is here at the wrong Place please close and  move.

Ravenor, 2025-07-06:
Energy_forward not used. Power, with 230V from config.ini, the current is calculated and set to L1 current and above. Voltage at L1,L2 L3 ignored. Power L2,L3 ignored. Current calculatd for L1, L2 and L3 ignored. Energy_forward ignored.  and in the Venos topic it is under Energy/Forward. At L1,L2,L3 and above.

mr-manuel, 2025-07-06:
Sample flow:

[image]

Code:

```json
[
    {
        "id": "bf49cbdd8908a237",
        "type": "inject",
        "z": "2967729a97ee8c0e",
        "name": "",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "2300",
        "payloadType": "num",
        "x": 90,
        "y": 940,
        "wires": [
            [
                "c91a3be4748f34ba"
            ]
        ]
    },
    {
        "id": "df41a22e9a20f61e",
        "type": "inject",
        "z": "2967729a97ee8c0e",
        "name": "",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "230",
        "payloadType": "num",
        "x": 90,
        "y": 1000,
        "wires": [
            [
                "d7b2afb02a7932bc"
            ]
        ]
    },
    {
        "id": "a14ed1db28dd233f",
        "type": "debug",
        "z": "2967729a97ee8c0e",
        "name": "debug 2",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
[... 247 lines cut ...]
                "tot": "flow"
            },
            {
                "t": "move",
                "p": "payload",
                "pt": "msg",
                "to": "payload.pv",
                "tot": "msg"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 590,
        "y": 860,
        "wires": [
            [
                "a14ed1db28dd233f"
            ]
        ]
    }
]
```

closed 2025-07-18 by mr-manuel