mr-manuel/venus-os_dbus-mqtt-pv #65: dbus pv @ vrm portal - power is shown under inverters but shown as ac load too

geimel, 2025-07-19:
Hi,
first I'll thank you for this great work. 
I have a problem with the dbus pv driver. I send the power over mqtt and in the remote console the inverter is shown with correct power values.
In the vrm portal the power is shown under pv-invertes (in my case L1 Power) but it is also shown under a AC Load (L1 Power).
I have 2 Fronius Devices, these a shown correct und pv-invertes and not under AC Load.
Have you an idea where the problem is?
Thanks a lot 
Ric

mr-manuel, 2025-07-20:
Unfortunately not. It's very unusual, that it's shown in two places at the same time.

geimel, 2025-07-20:
[image]

Here you can see the problem. On the left side, you see the 2 Fronius inverters + the GWTT PV Device. On the right side ist the same, but when GWTT PV = 0W.

I've tested the [venus-os_dbus-mqtt-grid](https://github.com/mr-manuel/venus-os_dbus-mqtt-grid) too, with grid and genset settings. The result is nearly the same on AC load. The meter isn't shown itself in VRM portal. Is it possible the add here the inverter-meter setting? On remote console, the pyhsical meter (rs485) has a setup node where you can set the different roles(i.e. Grid meter, PV Inverter, Generator, AC Load and EV charger). The simulated Meter has not these setup node.

Thanks a lot and have a nice sunday
Ric

mr-manuel, 2025-09-29:
I think the best is you ask in the Victron Community, since it seems more a complete system related config issue. You could try to play with the position parameter first.

closed 2025-09-29 by mr-manuel